dubstar-04/Design #121: Unable to save empty drawing

Brett-Parker, 2023-02-09:
**Describe the bug**
When saving a new document with nothing in the canvas, error is thrown
 - Version [e.g. 43-aplha2]

**Step to reproduce an issue **
Steps to reproduce the behaviour:
1. Open application new
2. Save drawing
3. See error in builder
4. Save dialog remains on screen for the user with no error message

**Expected behaviour**
Should be able to save drawing without anything in.

**Supporting documents**
[image]

Brett-Parker, 2023-02-09:
#29 , #54 are related to this.

dubstar-04, 2023-02-10:
Need to remove the check for an empty canvas too:

https://github.com/dubstar-04/Design/blob/main/src/Design/window.js#L180

closed 2023-02-14 by dubstar-04