dubstar-04/Design #139: Layer colour is not perserved on opening for saved file

jaisekjames, 2023-02-11:
**Describe the bug or feature**
Layer colour is not perserved on opening for saved file
 - Version [e.g. 43-alpha3] - Check the about window under the app menu

dubstar-04, 2023-02-11:
Thanks for taking the time to make a bug report. Really appreciated.

dubstar-04, 2023-02-13:
This issue is a problem with converting the colour from RGBA to HEX to ACI (AutoCAD Colour Index)

The colours are selected using a Gtk.ColorButton: https://docs.gtk.org/gtk4/class.ColorButton.html
on the layers window: https://github.com/dubstar-04/Design/blob/main/src/Design/layersWindow.js#L120

Unfortunately, the colours provided by the GTK colour button don't always have an equivalent ACI colour so the colour can't be converted, resulting in the LAYER element being saved to the DXF file with a layer value (62) of 'undefined'. 

When opening a file the colour is assigned to white and later overwritten with the correct colour. In this case there is no correct colour so the layer remains white

dubstar-04, 2023-02-13:
I am not aware of a way to provide a custom colour set to a ColourButton (depreciated) or a ColorDialogButton
https://docs.gtk.org/gtk4/class.ColorDialogButton.html

This may require a colour picker to be created that only allows selection of ACI colours.

dubstar-04, 2023-02-13:
A good shortterm solution would be to use a custom colour pallet on a ColourChooser. Again these are depreciated. 

https://docs.gtk.org/gtk4/method.ColorChooser.add_palette.html

dubstar-04, 2023-02-15:
Related to #145