dubstar-04/Design #140: Unable to open DXF (nothing seems to be displayed)

Peque, 2023-02-11:
**Describe the bug**

When trying to open [this DXF file](https://drive.google.com/file/d/19o9hnaSCupBDQphN2hdyFg2oHEg7VOpi/view?usp=share_link) nothing seems to be displayed but no error message is shown either.

Note the file has a linked raster image for one layer. Visualizing this would definitely be a plus, but this issue aims to report only the "nothing is displayed" issue (the expected behavior would be to show the different layers with outlines/fills). :blush: 

 - Version 43-alpha3

dubstar-04, 2023-02-11:
This is excellent. Thanks for taking the time to make a bug report.

Peque, 2023-02-11:
Thanks to you for sharing your work! :blush:

dubstar-04, 2023-02-13:
Found the issue: 

[image]

dubstar-04, 2023-02-13:
@Peque The reported issue should now be fixed. Please continue to report any issues, helping to improve Design.

closed 2023-02-13 by dubstar-04

Peque, 2023-02-13:
@dubstar-04 Awesome! :blush: 

Should I open an issue in this project for all layers being white and no fill color displayed? Or is it enough with https://github.com/dubstar-04/Design-Web/issues/155 ? (I'm guessing it is the same issue, but maybe not)

dubstar-04, 2023-02-13:
I had a look at your file and all the colours are list as ACI 7
Where is the file being exported from?

There is a similar issue here for layers:
https://github.com/dubstar-04/Design/issues/139

The hatching and fill isn't currently supported. Please feel free to open a bug for that.

Peque, 2023-02-13:
@dubstar-04 I opened this issue with more information: https://github.com/dubstar-04/Design/issues/145

commit 08d5c2d9 (2023-02-13): Merge pull request #144 from dubstar-04/Update-core

Update Design-Core. Fixes #140
commit 1e62aead (2023-02-13): Update Design-Core. Fixes #140