dubstar-04/Design #145: Add support for True Color (DXF attribute `420`)

Peque, 2023-02-13:
References:

- [AutoCAD reference searching for `420`](https://help.autodesk.com/view/OARX/2018/ENU/?query=420)
- [Adding layers with True Color and AutoLISP](https://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-F392F4F2-B2C1-4C8E-AEBB-1D3FE31610C9)
- [Common Group Codes for Entities (DXF)](https://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-3610039E-27D1-4E23-B6D3-7E60B22BB5BD)

The issue can be reproduced by opening [this DXF file](https://drive.google.com/file/d/19o9hnaSCupBDQphN2hdyFg2oHEg7VOpi/view?usp=share_link).

As an example, the "Ground" layer has the True Color (`420`) defined as:

```
Ground
 70
0
 62
7
420
15921906
```

Which corresponds to `0xF2F2F2`.

Another example would be the "Roof" layer, which uses a yellow color (`0xEACA3E`) represented as `15387198`.

dubstar-04, 2023-02-14:
This is excellent. Thank you.

dubstar-04, 2024-01-20:
[image]

closed 2024-01-20 by dubstar-04

Peque, 2024-01-20:
@dubstar-04 Looks great! Will try it with the next Flathub update. :blush:

dubstar-04, 2024-01-20:
> @dubstar-04 Looks great! Will try it with the next Flathub update. 😊

Next release will probably be March with GNOME 46. I'm trying to get as many fixes in before then and I can and possibly add dimensions too.

Peque, 2024-01-20:
I'll be keeping an eye then. Thanks for sharing! :muscle: :blush:

Peque, 2024-06-07:
Working great after latest Flatpak upgrade. :rocket: :blush: