dubstar-04/Design #229: Ifc support thru https://ifcjs.io/

Isaina, 2023-07-29:
Import ifc into dxf with floor plan and floor browser and save floor plans to dxf for editing.

Isaina, 2024-01-07:
Yes dxf export is great but Import IFC and floor browser would be great with export to dxf and blocks support would give this software a purpose , so we can import 3d ifc model make 2d floor plan and export to 2d dxf file

dubstar-04, 2024-01-07:
This is a great idea but out of scope for this project. Design is aiming to follow the GNOME philosophy of 'do one thing well'. 

The goal is to be the simplest drawing editor possible, focussing in DXF files. With that being said I am aiming to make the core scriptable this year so it would be possible to import other file types via extensions. 

I am going to close this feature request for now but may revisit the idea once extensions are possible.

closed 2024-01-07 by dubstar-04

Isaina, 2024-01-07:
I was thinking thru https://github.com/IFCjs/components

This library is a collection of BIM tools based on [Three.js](https://github.com/mrdoob/three.js/) and other libraries. It includes pre-made features to easily build browser-based 3D BIM applications, such as postproduction, dimensions, floorplan navigation, DXF export and much more.

not 3d but to extract 2d from ifc and save to dwg