dubstar-04/Design #230: Export to pdf

Isaina, 2023-07-29:
Export selection to pdf with page size like a1;a2;a3;a4

dubstar-04, 2024-01-07:
Are you more interested in paper space or model space prints?

Isaina, 2024-01-07:
The easiest would be model space with window selection but also can be and paper space just to select paper size and size ratio or fit to page

jaisekjames, 2024-01-29:
Add multi-page pdf output. 
Open multi plot
Select Outer borders [polyline/block] in model space in required order. [say 10 dwg will have 10 borders. So it will generate 10 pdf concatenated output].

Please refer this for more detail
https://www.progesoft.com/products/progecad-professional/manual?mp=progecad-features/progecad-extra-functions/multi-plot

dubstar-04, 2024-01-29:
Thank you for the input on this feature request. Knowing which features are most important help me decide what to focus on. It's also nice to know that other care about this project. ❤️

gianlucafiore, 2025-09-04:
Hi, I just found this project. I'm an Architecture student almost graduating, and an amateur JS developer.

This is great! When I have time, I'd be interested in contributing code and features. I think it would be great to be able to replicate the features offered by proprietary software and leave the API completely open, allowing the software to be as modular as possible, so the community can contribute plugins, macros, and tools!

dubstar-04, 2026-10-08:
Added PDF and SVG plotting with #470

closed 2026-10-08 by dubstar-04