dubstar-04/Design #231: Dark mode/theme does not apply to drawing area background

mardab, 2023-09-14:
**Describe the bug or feature**
A clear and concise description.
 - Version 44-alpha4 [flatpak]
 
Upon changing global theme for GTK apps, only top and bottom bars have the colors applied, while drawing area remains white.

**Step to reproduce an issue (omit for feature request)**
Steps to reproduce the behaviour:

1. set dark theme in GTK applications theme settings or start application with valid theme variable (e.g. `GTK_THEME=Adwaita:dark`)
2. open Design

**Expected behaviour**
Drawing space background colors should match selected GTK theme.

**Supporting documents**
[image]

dubstar-04, 2023-09-15:
Thanks for the bug report. This functionality has been tested and works on fedora and Debian. Which is are you using? There is something strange going on with your setup as all the fonts and spacing looks different.

WilliamTCarroll, 2023-09-28:
I would like to chime in; I have the same issue.  I am using Linux Mint with the Mate Environment instead of Gnome (it is a nice, clean environment that was forked from Gnome 2).

I am on `44.alpha4` via flatpak.

Would further info might be helpful?

Perhaps a possible work-around could be an option to change the theme in the "Preferences" menu?  Light, Dark, or Defer to OS?

dubstar-04, 2024-06-07:
Is this still an issue with the latest version Design 46-alpha1? If the problem persists I will have to add the theme to the preferences like you suggest.

mardab, 2024-06-10:
In the meantime I have switched distribution, reloaded similar setup in KDE 5.27 and this is how `46-alpha1` looks like **without** manually switching GTK theme:
[image]