dubstar-04/Design #232: Cannot open any file

williamd1k0, 2023-09-24:
I cannot open any file.

Version: 44-alpha4 and main

Steps to reproduce the behaviour:
Just try opening a file using Open Dialog or command-line argv.

## If using Open Dialog
The dialog itself simply doesn't appear.
After messing with the code, I managed to get it to work by setting the file filter after the dialog initialisation.

```diff
diff --git a/src/Design/fileIO.js b/src/Design/fileIO.js
index e3c7789..1964394 100644
--- a/src/Design/fileIO.js
+++ b/src/Design/fileIO.js
@@ -25,11 +25,11 @@ export class FileIO {
 
     const dialog = new Gtk.FileChooserNative({
       action: Gtk.FileChooserAction.OPEN,
-      filter: filter,
       select_multiple: false,
       transient_for: window,
       title: _('Open'),
     });
+    dialog.add_filter(filter);
 
     dialog.show();
     dialog.connect('response', (dialog, response) => {
```

## If using command-line argv
The app crashes before opening.
The log will warn you of this error:
```js
JS ERROR: TypeError: activeWindow.load_file is not a function
DesignApplication/</<@resource:///io/github/dubstar_04/design/js/Design/main.js:73:26
```
I haven't found a way to fix it myself though.

dubstar-04, 2023-09-25:
Thanks for the report. This works for me on Debian and Fedora. File opening is done through the flatpak portal. 
What OS are you using?

williamd1k0, 2023-09-26:
I'm using Manjaro 23.0.2 Xfce4.

dubstar-04, 2023-10-09:
@williamd1k0 please can you provide examples of the files that won't open?

dubstar-04, 2023-10-09:
How are you launching the application?

dubstar-04, 2024-01-05:
No response. Closing issue.

closed 2024-01-05 by dubstar-04