dubstar-04/Design #234: ReferenceError: Logger is not defined

Peque, 2023-10-17:
The current version available in Flatpak seems to be unable to open [this file](https://github.com/dubstar-04/Design/files/12961869/plan.txt)  (note you'll need to rename the extension from `.txt` to `.dxf`):

[image]

@dubstar-04 Can you reproduce it?

dubstar-04, 2023-12-30:
I can't reproduce the exact error. I get a different error about invalid line types. Thank you for supplying the file and reporting the issue.

Peque, 2023-12-30:
In case it helps, I installed the latest Flatpak package from Flathub using Fedora 39. I guess the OS should not matter, but just in case there is some reference not properly set an using some OS Logger instead of the one that should be available in the runtime.

dubstar-04, 2024-01-07:
Thanks for reporting this issue. 

I think it is now fixed:

[image]

closed 2024-01-07 by dubstar-04

dubstar-04, 2024-01-07:
> Thanks @dubstar-04 . The same error persists. 🤷

Are you building a flatpak from main?

Peque, 2024-01-07:
@dubstar-04 No; just used the latest Flatpak version available in Flathub. I have just noticed the app has not been updated there yet, so you can ignore my comment. I will re-test once there is an update in Flathub. :blush:

dubstar-04, 2024-01-07:
Phew!

Peque, 2024-06-07:
Fixed after latest Flatpak upgrade. :rocket: :blush: