dubstar-04/Design #248: Latest git checkout fails to build

prokoudine, 2024-01-28:
The latest code (commit 75511204) fails to build on Ubuntu 23.10:

```
[1/8] Generating src/io.github.dubstar_04.design.src_gresource with a custom command
FAILED: src/io.github.dubstar_04.design.src.gresource 
/usr/bin/glib-compile-resources ../src/io.github.dubstar_04.design.src.gresource.xml --sourcedir ../src --internal --generate --target src/io.github.dubstar_04.design.src.gresource --dependency-file src/io.github.dubstar_04.design.src.gresource.d
../src/io.github.dubstar_04.design.src.gresource.xml: Failed to locate “Design-Core/core/designCore.js” in any source directory.
[6/8] Generating data/blueprints with a custom command
ninja: build stopped: subcommand failed.
```

Any additional info I can provide you?

dubstar-04, 2024-01-28:
The error suggests that design-core submodule isn't updating. You could try a fresh clone or git pull --recurse-submodules

dubstar-04, 2024-01-28:
If this is a fresh clone you will need to check if the Design-core directory has any contents. If you may need to clone with submodules: git clone --recurse-submodules

prokoudine, 2024-01-29:
Silly me, sorry about the noise.

closed 2024-01-29 by prokoudine

dubstar-04, 2024-01-29:
No worries. It's nice to know that someone is interested enough to be playing and testing.